hsgw/plaid #5: BOM Parts

mrceephax, 2019-04-10:
Hi, 

I would be so helpful to have the parts organised either through octopart.com or with some serial numbers, I'm able to find them but its a bit difficult when the "Value" term is used as a generic  attribute.

Thanks

imbarelyonline, 2019-04-16:
I used Octopart to keep track of components as I found them on Mouser. You can use (and correct!) my list if you'd like. [Public Plaid Electronics](https://octopart.com/bom-tool/t5azwMqr/G5hAETgBxOPJR9aY)

miguelangelramirez, 2019-04-24:
Don't know if you guys have stumbled into this link to octopart with the list: https://octopart.com/bom-tool/t5azwMqr/G5hAETgBxOPJR9aY

lscott3, 2020-08-13:
So it looks like the two links shared here are the same.

Can someone verify that all of the parts listed work? If so, I think I'll open a PR to add the more detailed and exact name in the BOM list.