hsgw/plaid #6: make plaid:default:program fails with avr-gcc 8

brianmutualaid, 2019-04-15:
I just ran into this issue on macOS and thought I would open an issue just in case others have it too. When using the default avr-gcc (8.2.0) that's installed with homebrew, I get the following output when running `make plaid:default:program`:

```$ make plaid:default:program
QMK Firmware 0.6.308
WARNING:
 Some git sub-modules are out of date or modified, please consider running:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making plaid with keymap default and target program

avr-gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  14666	      0	  14666	   394a	.build/plaid_default.hex

Creating load file for EEPROM: .build/plaid_default.eep                                             [OK]
Copying plaid_default.hex to qmk_firmware folder                                                    [OK]
Checking file size of plaid_default.hex                                                             [OK]
 * The firmware size is fine - 14666/32768 (18102 bytes free)

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ".build/plaid_default.hex"
avrdude: input file .build/plaid_default.hex auto detected as Intel Hex
avrdude: ERROR: address 0x8002a7 out of range at line 920 of .build/plaid_default.hex
avrdude: read from file '.build/plaid_default.hex' failed

avrdude: safemode: Fuses OK (E:FC, H:D0, L:D7)

avrdude done.  Thank you.

make[1]: *** [program] Error 1
make: Make finished with errors
*** [plaid:default:program] Error 1```

I found a similar issue mentioned [on Reddit](https://www.reddit.com/r/MechanicalKeyboards/comments/apflr6/ergotravel_error_flashing_qmk_firmware_error/) and was able to work around it in the same way, but installing avr-gcc version 7.

Maybe this would be worth mentioning in the [Firmware](https://github.com/hsgw/plaid/blob/master/doc/en/firmware.md) section of the docs? I'd be happy to add a section for it and open a pull request, just let me know if that would be helpful.

hsgw, 2019-04-15:
Thank you for letting me know.

Please send PR!

brianmutualaid, 2019-04-15:
Opened a PR [here](https://github.com/hsgw/plaid/pull/7).

hsgw, 2019-04-16:
Thank you so much!

closed 2019-04-16 by hsgw