hsgw/plaid #9: Trying to open .chm file on Kicad, symbols display question marks

miguelangelramirez, 2019-04-24:
Symbols seem to be not displaying correctly, I'm using Kicad 5.1 BTW. Thanks for the files too!

nicholasterr24, 2019-05-02:
Hey, I am almost a complete beginner at keyboards, KiCad, and electronics, but I think this is a fix for most of the symbols. When you double click them on the schematic go to the Symbol library and type in whatever the symbol is supposed to be and find the correct library. For example, for the USB connector on the top left of the schematic, it says conn: USB_OTG, so i click on the browse library button and look up USB_OTG and it gives me the correct symbol. A lot of them also say "device: [something]" and you just have to change them to Device: [something] with a capital D instead of device: [something]. The only symbol I have not found so far is the ATMEGA328-PU or the GS2 symbols.

hsgw, 2019-05-07:
Hi,

As @nicholasterr24 says, please collect footprint library name.

I think that the cause of this is changing kicad footprint name.
So I should fix this soon.

Thanks.

hsgw, 2019-05-07:
I fixed this issue.   
Can anyone try this schematic file?
https://github.com/hsgw/plaid/tree/kicad_sch

nicholasterr24, 2019-05-07:
Yeah, I just opened it and everything seems to look good, all of the symbols show up correctly

miguelangelramirez, 2019-05-07:
Hello all, @nicholasterr24 and @hsgw. Got around to fix it during the weekend using the same method double clicking the symbols. THANKS!

bjvanst, 2019-06-11:
> 
> 
> I fixed this issue.
> Can anyone try this schematic file?
> https://github.com/hsgw/plaid/tree/kicad_sch

That version of the project loads the symbols automatically for me in KiCad 5.1.2.

matthewwiese, 2019-12-26:
> I fixed this issue.
> Can anyone try this schematic file?
> https://github.com/hsgw/plaid/tree/kicad_sch

This also fixed my "??" missing symbol problem as well — I am using KiCad version 5.1.5

It seems like a fix to me. Should this be merged with master and the issue closed?