hsgw/plaid #10: Help with build

miguelangelramirez, 2019-06-06:
Hey guys, I managed to solder all the components into the PCB including switches. Currently trying to burn the bootloader from my mac but have had no luck just yet. 

Should LEDs turn on once connected?

I've used the direction in the bootloader.md file and also in the _alpha file with no luck :(

hsgw, 2019-06-06:
Hi,

LED is not turn on when connected.

Could you check soldering and the directions of component again?

miguelangelramirez, 2019-06-06:
Double checked the soldering, thought I saw a bridge on the capacitor but there was no bridge at all
[image]

BenRoe, 2019-06-06:
Are you using the ISP header? Which device you use to burn the bootloader on the chip?
Check also https://github.com/baerwolf/USBaspLoader/issues/19

miguelangelramirez, 2019-06-06:
I'm plugging the keyboard to my computer through USB, should I use the ISP header to burn the bootloader??

BenRoe, 2019-06-06:
Do you bought the kit via the original [GB](https://www.reddit.com/r/mechmarket/comments/apf0ng/gb_plaid_throughhole_kit_12x4_ortholinear/) or [keycapsss.com](https://keycapsss.com)?
If yes, the bootloader is already burned and you only have to follow the[ firmware steps.](https://github.com/hsgw/plaid/blob/master/doc/en/firmware.md)

miguelangelramirez, 2019-06-06:
I ordered my PCB from jlcpcb since I couldn't get a kit, I know I suck.

LorenzoMarsicano, 2019-06-06:
I've bought the kit from the GB, but even after another check to both soldering and schematics my pc (both windows and Linux) won't notice that a new device has been connected to it.

Could it be that the atmega is not working properly? 
How do I check that?

BenRoe, 2019-06-06:
Ok, i can give you some tips, but can't explain the whole process.
You need a ISP Flasher like the usbtinyisp (cheap on ebay or aliexpress).
Connect the usbtinyisp via usb to the pc.
Connect usbtinyisp to plaid via isp header.
Use avrdudess (gui for avrdude) on windows. It's easier.

Use yellow marked settings, but replace it for the following settings.
L (LFUSE): 0xF7
H (HFUSE): 0xD0
E (EFUSE): 0x04
[image]

LorenzoMarsicano, 2019-06-06:
Wow, thank you very much, this was way more than I expected from you!

I'll try this as soon as possible, thanks!

miguelangelramirez, 2019-06-06:
> Ok, i can give you some tips, but can't explain the whole process.
> You need a ISP Flasher like the usbtinyisp (cheap on ebay or aliexpress).
> Connect the usbtinyisp via usb to the pc.
> Connect [...]

That is awesome!! sadly they take some time to delivery and I'm moving to LA soon :(. I'll order one when I get my new address, but still try to load it with my Arduino Uno/breadboard...I think I have all the parts just laying round. 

Thanks bro!

LorenzoMarsicano, 2019-08-07:
I've tried flashing the bootloader with the USBTiny programmer with no luck at all.

The most weird thing is that avrdude is successful in writing the bootloader even if the board isn't officially in DFU mode (success either after or before pressing the buttons).

itsnoteasy, 2020-02-09:
here's how I got it to work.
1, open plaid_default.hex in text editor, remove last line :00000001FF
2, append usbasp bootloader on to end
3, flash using teensy 2.0 with qmk isp flashing guide
avrdude -c avrisp -p m328p -P /dev/ttyACM0 -b19200 -U flash:w:donegal_default.hex
avrdude -c usbasp -b19200 -p m328p
one fuse didn't write but it wasn't important.
then make plaid:default:program to program normally over usb.
the issue seems to be that an incorrect baud rate was used.

BenRoe, 2020-02-09:
If you use a newer qmk firmware, then use `make plaid:default:flash` and not `plaid:default:program`.

itsnoteasy, 2020-05-28:
I was having difficulty entering the bootloader and i'd like to note the docs are wrong. the correct procedure is:
Enter bootloader mode

    Push and hold BOOT SW
    Plug into USB cable
    Release BOOT SW
    make plaid:flash
    Push and release RESET SW

denisinvader, 2020-08-28:
> I was having difficulty entering the bootloader and i'd like to note the docs are wrong. the correct procedure is:
> Enter bootloader mode
> 
> ```
> Push and hold BOOT SW
> Plug into USB cable
> Release BOOT SW
> make plaid:flash
> Push and release RESET SW
> ```

This worked for my kit ordered from mechboard (preflashed I guess)

Daveyr, 2021-01-06:
@itsnoteasy, would you mind elaborating? Are you using the `main.hex` file from the plaid fork of the USBaspLoader repository? In step 2, which file contents do you append to the bottom of it? I'm trying to use avrdudess in Linux (using mono), with an Arduino nano as a programmer but not having any success. My settings are below but I think the main issue is the hex file.
[image]

itsnoteasy, 2021-01-06:
that's correct, step 2 is copy paste the ascii from this file https://github.com/hsgw/USBaspLoader/blob/plaid/firmware/main.hex
I don't know what the error is with your setup but it could be using the nano for programming isp. a teensy 2.0 would have a lower learning curve.

Daveyr, 2021-01-09:
Thanks for the clarification. So I combine the main.hex from [here](https://github.com/hsgw/USBaspLoader/blob/plaid/firmware/main.hex) with the main.hex from your file above into one file (removing the last line from the first file - contains `:00000001FF`)?
I've named this plaid_bootloader.hex and managed to upload it using an arduino nano using the following command:

```
avrdude -u -c arduino -p m328p -P /dev/ttyUSB0 -b 115200 -U flash:w:"/home/ubuntu/Documents/projects/keyboards/plaid/bootloader/plaid_bootloader.hex":a -U lfuse:w:0xF7:m -U hfuse:w:0xD0:m -U efuse:w:0x04:m
```
However, it complains about the efuse setting.

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0x04:
avrdude: load data efuse data from input file 0x04:
avrdude: input file 0x04 contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xfc != 0x04
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Incidentally, this is exactly the same error I find when I burn the bootloader using Arduino software. I created a new board definition with the fuse settings you showed above and the error message was almost identical.

When I run `avrdude -c usbasp -p m328p` it says that no such device is connected. Any ideas?

Daveyr, 2021-02-01:
After much trial and error I think I've worked it out. I wrote a blog post [here](https://www.algorist.co.uk/post/how-to-burn-the-plaid-keyboard-bootloader-using-arduino-nano/) but the summary is below.

First upload the standard avrisp example sketch to the Arduino Nano. Note, the default baud rate is 19200.
Using the method described [above](https://github.com/hsgw/plaid/issues/10#issuecomment-583849113) to make the hex file, I entered the following in a terminal on a pc connected to the Arduino, in turn connected to the ISP headers on the keyboard.

```
avrdude -u -c avrisp -p m328p -P /dev/ttyUSB0 -b 19200 -U flash:w:"plaid_default.hex":a -U lfuse:w:0xF7:m -U hfuse:w:0xD0:m -U efuse:w:0xfc:m
```
Notes.
1. When burning the bootloader you need a 10nF capacitor across GND and RST pins on the Arduino Nano (no need for other Arduino types)
2. The programmer is actually `avrisp` rather than arduino
3. The efuse setting in the plaid firmware repo is wrong; it should be 0xfc
4. I had to disconnect the Arduino from the keyboard in order for it to be properly detected

MrCrazyUkrainian, 2021-05-27:
Hi all!
I am having the same problem as originally described in this thread: I soldered everything seemingly correctly and yet the keyboard isn't detected by my WIndows PC in the device manager. It just lists it as an unrecognized USB device. I followed the guide and installed the libusBK driver and was using button press combinations from the OG guide and what was reccomended by @denisinvader , and yet it stil isn't recognized. Does anyone of you have any ideas what I should do?
I bought the kit from mechboards.co.uk and it is supposed to have firmware preflashed onto the ATmega ship.

chief-teep, 2021-06-04:
Was the board in bootloader mode when you installed libusbk in zadig? It should show up as usbasb for the device if done correctly.

I made the mistake of installing libusbk as the device driver instead of the bootloader and had the same issue as you did.

MrCrazyUkrainian, 2021-06-04:
Update on my troubles. They are now resolved.
Solder USB connector carefully and check multiple times that all the pins have continuity to the corresponding components, as in my case one of the data lines wasn’t soldered properly (solder pad split) and that was the cause of everything.
After plonking a bit of solder onto that problematic joint everything works fine.
Admittedly it took me and MechBoards UK support staff way too much time to figure out.

chief-teep, 2021-06-04:
Glad you got it sorted! Don't feel too bad about it taking so long to figure out, I flashed a keymap to my plaid using the newest qmk version and it was bricked for 2 months before I figured out that I needed to use an older release -_-

Ardakilic, 2022-04-28:
Late to the party. I could not program the chip with USB ISP programmer, and I ended up using a Raspberry Pi instead. The All of my programmers, although they showed uploaded successfully (tried on macOS), did not apparently upload the bootloader to my cheap atmega328p.

The issue for me was both for Plaid and Lumberjack, which apparently is quite similar in terms of technicality.