hsgw/plaid #11: Cannot enter bootloader mode

nguyenhimself, 2019-07-16:
I had my Plaid working for a month. Now I want to flash a new layout.
I understand that to enter bootloader mode, I have to:

1. Plug into USB cable
2. Push and hold RESET SW
3. Push and hold BOOT SW
4. Release RESET SW
5. Release BOOT SW

However, when I pushed the BOOT button, it just output "edc", i.e the keys on the 4th column, so now I cannot enter bootloader mode to flash my new layout.
Again, aside from the BOOT button, all the other keys of my Plaid still works fine.
How should I go about solving this? Thanks.

Snowdrifft, 2019-07-16:
check if your RESET button works, it will disconnect and reconnect the board.
Will return "edc" if switch button is not pushed down while pressing the BOOT button

miguelangelramirez, 2019-07-16:
Cool, just ordered a new atmega chip with a pre-loaded bootloader on it. Will check on this method when I finally get it. Thanks!

> check if your RESET button works, it will disconnect and reconnect the board.
> Will return "edc" if switch button is not pushed down while pressing the BOOT button as BOOT switch is using the same pin as COL3 (See schematics).

closed 2019-08-02 by hsgw