hsgw/plaid #13: Running "make plaid:default" fails (Error 128)

rjth, 2019-08-01:
Running _make_ results in the following errors on Windows:

```
PS C:\dev\qmk_firmware> make plaid:default
      0 [main] sh 12400 sync_with_child: child 10744(0x1BC) died before initialization with status code 0xC0000142
    106 [main] sh 12400 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
      0 [main] sh 5636 sync_with_child: child 10812(0x1C8) died before initialization with status code 0xC0000142
     94 [main] sh 5636 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
      0 [main] sh 15020 sync_with_child: child 8084(0x1B8) died before initialization with status code 0xC0000142
     94 [main] sh 15020 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
      0 [main] sh 568 sync_with_child: child 13728(0x1B4) died before initialization with status code 0xC0000142
    129 [main] sh 568 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
make: *** [plaid:default] Error 128
```

I had no issues entering bootloader mode or getting USBasp recognized. I have pulled the latest qmk firmware package from git. 

Any help is appreciated.

hsgw, 2019-08-02:
Hmm, this is so weird.
Maybe this is caused by build environment.

Could you try to make other keyboard's firmware like this.
```
make planck;default
```

rjth, 2019-08-02:
Thank you for the quick response. Unfortunately I will be only able to test it tomorrow. I will get back here with the results.

rjth, 2019-08-02:
It looks like all make commands are failing the same way with QMK. It is probably unrelated to Plaid. I will try to look into it.

BenRoe, 2020-01-06:
Do you solved the problem?
If yes we can close this issue.