hsgw/plaid #15: Wrong keys being sent.

ekosynth, 2019-08-24:
[image]

After building the Plaid all the keypresses were registering correctly, except one dead key. I was trying to trace this but since then only some of the keypresses work, and they dont match the keys that I'm pressing. A lot of the keys seem to trigger some combination with ctrl and alt too.

I've tried reflashing the firmware, but nothing seems to affect it. Any idea what I've done wrong?

hsgw, 2019-08-26:
Please check switch matrix isn't short.
And test this keymap below.
```
[0] = LAYOUT_plaid_grid(
    KC_A,     KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,
    KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,
    KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,
    KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO
),
```

ekosynth, 2019-08-26:
@hsgw I tried the keymap you sent and this is what I get pressing all the different keys:
[image]

I can't seem to find a short in the matrix. I wonder if maybe I've damaged the atmega328?

Bauncer, 2019-09-24:
I’ve just had the exactly same problem with my plaid yesterday. Rebooted my PC and keypresses were messed up. Found this in r/olkb: https://www.reddit.com/r/olkb/comments/d35aqo/help_with_plaid_thk_sending_seemingly_random/

Will try the fix later

gipsy-king, 2020-08-14:
confirming that
```void matrix_init_user(void) {
  eeconfig_init();
}```
also fixed the issue for me.

elasticrash, 2020-09-12:
I am not quite certain if its related. I had a similar issue. Only when I press the key where the spaces are, it then switched to layer 3. I flashed an option on layer 3 to go back to layer 1 and that works. But I have to do it every time I plug it in. I will try the matrix init and see if that makes a difference