hsgw/plaid #20: Unclear documentation

CrisHale, 2020-06-23:
In the file https://github.com/hsgw/plaid/blob/master/doc/en/bootloader.md it says "If you buy kit, already burn bootloader in ATMEGA328p" but it's not clear what this means, or what you should do if you bought a kit.

The section for what to do with Windows is also a bit simple. It doesn't give you any clues on how to use Zadig. The software suggests the USB item should be plugged in and chosen from a list, but in the Plaid documentation you're not asked to plug in the USB cable until the next step (Enter bootloader mode).

BenRoe, 2020-06-28:
@CrisHale the plaid and plaid gold bootloader and firmware are identical.
Here you can find more information about how to flash the bootloader with avrdudess https://github.com/baerwolf/USBaspLoader/issues/19

CrisHale, 2020-06-29:
@BenRoe Thanks, I will try restoring the correct bootloader on the chip I have.

BenRoe, 2020-06-29:
@CrisHale do you changed the bootloader, or bricked the mcu?
The Plaid Gold kit comes with a pre flashed bootloader and firmware. So everything should work right after correct soldering.

CrisHale, 2020-06-29:
@BenRoe I don't know if you remember me, but I'm the guy who soldered the USB connector on upside-down. I flashed a new bootloader before I realised that was the real problem. I assume the chip still works, but I can't test it until the replacement kit you've sent gets here and I solder it up (properly this time).
I was just thinking that if I ever do manage to fix the soldering, I could try the PCB I have again to see if it works.

BenRoe, 2020-06-29:
@CrisHale yes i remember. Hope the replacement arrives soon.

@hsgw you can close this issue

closed 2020-06-29 by CrisHale