hsgw/plaid #22: Update Files for USB-C Support

Tenkir, 2020-06-30:
Currently the PCB is laid out for micro-USB.

hsgw, 2020-07-03:
Ah, I think USB-C through hole connector is little unusual. So I would like to keep it mini-B.

Tenkir, 2020-07-09:
I've seen these available, have never tried them though.
https://www.newark.com/edac/690-024-632-031/usb-connector-3-1-type-c-rcpt/dp/98Y3918

Tenkir, 2020-07-18:
Just found this kit which uses a through-hole USB-C connector: 
https://keycapsss.com/diy-kits/104/plaid-pad-4x4-macro-pad/numpad

piit79, 2020-08-08:
Both Discipline (65%) and Mysterium (TKL) through-hole boards use a through-hole USB-C connector: https://www.digikey.com/product-detail/en/USB4085-GF-A/2073-USB4085-GF-ACT-ND/9859733 I'll take a stab at modifying the PCB for this connector.

piit79, 2020-08-11:
So, looks like I managed to do it, see the pull request above. I included the zip file with the gerbers if anybody wants to inspect them. This was my first go at KiCad, so I hope I haven't messed anything up. Guess I'll have to order some prototypes and see for myself :)

bhuhnke, 2020-08-12:
I'd love to see this implemented! Maybe as a separate USB-C version?

piit79, 2020-08-12:
@bhuhnke Yes, I was also thinking a separate branch would be warranted because of the limited availability of the through-hole USB-C connector (I only found it at DigiKey...).

Either a new branch for the USB-C version, or make the USB-C default and create a new branch for the current Mini USB version.

bhuhnke, 2020-08-12:
> @bhuhnke Yes, I was also thinking a separate branch would be warranted because of the limited availability of the through-hole USB-C connector (I only found it at DigiKey...).

I‘ve got a spare one around from a self ordered discipline built and I‘d love to try it on the new pcb! Uploaded the gerber file to jlcpcb just for kicks and they didn‘t spit out any errors. I don‘t know if they check anything with the files while uploading though.

piit79, 2020-08-12:
@bhuhnke I don't suppose you're in the EU? I'm probably going to order the minimum quantity which is 5 from JLCPCB, could easily send you one for testing.

bhuhnke, 2020-08-12:
@piit79 I‘m from germany :) That would be awesome!