hsgw/plaid #26: What are the diameters for the screws?

CrisHale, 2020-07-02:
I have a screw with a bad thread (one of the long ones). I can make do with three for now, but I'll see how much to replace it if I can find out the diameter, or specification generally, for it. Thanks.

hsgw, 2020-07-03:
Hi, Are you talking about the spacer for acrylic plate?
If so, they are 8mm.

CrisHale, 2020-07-04:
Hi, the screw itself. The 8 mm ones, yeah. I need the diameter as well, and possibly other information. Or a link to where they come from would do nicely, if you've got a source. Thanks.

piit79, 2020-08-13:
@CrisHale The screws are M2 (see [BOM](https://github.com/hsgw/plaid/blob/master/doc/en/BOM.md)), so the diameter is a bit less than 2 mm (see [ISO metric screw thread](https://en.wikipedia.org/wiki/ISO_metric_screw_thread) on Wikipedia).