hsgw/plaid #30: Connect OLED?

denisinvader, 2020-09-18:
I saw [this post](https://www.picuki.com/media/2229336102125338862) but couldn't find any info.

I would like to connect OLED screen (SSD1306) and have some questions:
1. Should I solder I2C resistors?
    - If I should, which value?
1. Will LEDs works the same way with OLED screen connected and controlled?

frstff, 2021-02-19:
Hi there! 
If You're still interested, here are the answers: 

1. No, all you need is to wire 4 I2C pins to corresponding holes on the board.
2. You'll have to desolder LED bridges on PCB to use I2C, therefore you won't be able to use them.
I think true to the spirit of the board solution would be to redesign the PCB for I2C ports to match those on standard OLED screen and connect it directly.

Ardakilic, 2021-07-16:
Hello, 

Any how-tos or guides available to make this mod?

Thanks,

frstff, 2021-07-16:
Hi!
I've just bought 4-pin connector for LED strips, cut it in half, checked which pin corresponds to which wire with multimeter, soldered it according to pins on OLED screen, and plugged the screen in.

Ardakilic, 2021-07-18:
Hey @frstff , thanks for the reply!

> checked which pin corresponds to which wire with multimeter,

My Plaid is still on the say (being printed). If the pins are shown with silkscreen, it'd be easy. VCC and GND identification should not be a problem. When PCBs arrive and I still can't manage I'll let you know. Thanks!