hsgw/plaid #33: Verification error when flashing bootloader.

Tenkir, 2020-11-19:
When attempting to flash `main.hex` from the [bootloader ](https://github.com/hsgw/USBaspLoader/blob/plaid/firmware/main.hex) repository, avrdude throws the following error:

```
$avrdude -u -c usbtiny -p m328p -P usb -b 115200 -V -e -U flash:v:"C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex":a 

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: erasing chip
avrdude.exe: verifying flash memory against C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
avrdude.exe: load data flash data from input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex auto detected as Intel Hex
avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex contains 30868 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x7000
             0xff != 0x6b
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

4dvn, 2020-11-19:
> When attempting to flash `main.hex` from the [bootloader ](https://github.com/hsgw/USBaspLoader/blob/plaid/firmware/main.hex) repository, avrdude throws the following error:
> 
> ```
> $avrdude -u -c usbtiny -p m328p -P usb -b 115200 -V -e -U flash:v:"C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex":a 
> 
> avrdude.exe: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.01s
> 
> avrdude.exe: Device signature = 0x1e950f (probably m328p)
> avrdude.exe: erasing chip
> avrdude.exe: verifying flash memory against C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
> avrdude.exe: load data flash data from input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
> avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex auto detected as Intel Hex
> avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex contains 30868 bytes
> avrdude.exe: reading on-chip flash data:
> 
> Reading | ################################################## | 100% -0.00s
> 
> avrdude.exe: verifying ...
> avrdude.exe: verification error, first mismatch at byte 0x7000
>              0xff != 0x6b
> avrdude.exe: verification error; content mismatch
> 
> avrdude.exe done.  Thank you.
> ```
> 
> If I then attempt to put the keyboard into bootloader mode, all USB devices on the system stop working until the keyboard is unplugged,

you need to set fuses and upload bootloader via avrdudess and use usbasp to upload

Tenkir, 2020-11-19:
> you need to set fuses and upload bootloader via avrdudess and use usbasp to upload

This is while attempting to install the bootloader via avrdude and usbtiny.

I am running these commands in debian:
```
make firmware
make flash
```

4dvn, 2020-11-19:
> > > When attempting to flash `main.hex` from the [bootloader ](https://github.com/hsgw/USBaspLoader/blob/plaid/firmware/main.hex) repository, avrdude throws the following error:
> > > ```
> > > $avrdude -u -c usbtiny -p m328p -P usb -b 115200 -V -e -U flash:v:"C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex":a 
> > > 
> > > avrdude.exe: AVR device initialized and ready to accept instructions
> > > 
> > > Reading | ################################################## | 100% 0.01s
> > > 
> > > avrdude.exe: Device signature = 0x1e950f (probably m328p)
> > > avrdude.exe: erasing chip
> > > avrdude.exe: verifying flash memory against C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
> > > avrdude.exe: load data flash data from input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex:
> > > avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex auto detected as Intel Hex
> > > avrdude.exe: input file C:\Users\tenkir\Downloads\USBaspLoader-plaid\USBaspLoader-plaid\firmware\main.hex contains 30868 bytes
> > > avrdude.exe: reading on-chip flash data:
> > > 
> > > Reading | ################################################## | 100% -0.00s
> > > 
> > > avrdude.exe: verifying ...
> > > avrdude.exe: verification error, first mismatch at byte 0x7000
> > >              0xff != 0x6b
> > > avrdude.exe: verification error; content mismatch
> > > 
> > > avrdude.exe done.  Thank you.
> > > ```
> > > 
> > > 
> > > If I then attempt to put the keyboard into bootloader mode, all USB devices on the system stop working until the keyboard is unplugged,
> > 
> > 
> > you need to set fuses and upload bootloader via avrdudess and use usbasp to upload
> 
> This is while attempting to install the bootloader via avrdude and usbtiny.
> 
> I am running these commands in debian:
> 
> ```
> make firmware
> make flash
> ```
> 
> ```
> .
> .
> ======>BUILDING BOOTLOADER FIRMWARE
> .
> make[1]: Entering directory '/home/jblair/git/USBaspLoader/firmware'
> make[1]: Nothing to be done for 'all'.
> make[1]: Leaving directory '/home/jblair/git/USBaspLoader/firmware'
> make[1]: Entering directory '/home/jblair/git/USBaspLoader/firmware'
> .
> avrdude -c usbtiny -p atmega328p -U flash:w:main.hex:i
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.01s
> 
> avrdude: Device signature = 0x1e950f (probably m328p)
> avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
>          To disable this feature, specify the -D option.
> avrdude: erasing chip
> avrdude: reading input file "main.hex"
> avrdude: writing flash (30866 bytes):
> 
> Writing | ################################################## | 100% 0.00s
> 
> 
> avrdude: error: usbtiny_send: error sending control message: Connection timed out (expected 128, got -110)
> avrdude: 30866 bytes of flash written
> avrdude: verifying flash memory against main.hex:
> avrdude: load data flash data from input file main.hex:
> avrdude: input file main.hex contains 30866 bytes
> avrdude: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 0.00s
> 
> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x7000
>          0x69 != 0x6b
> avrdude: verification error; content mismatch
> 
> avrdude: safemode: Fuses OK (E:FC, H:C0, L:1F)
> 
> avrdude done.  Thank you.
> 
> make[1]: *** [Makefile:41: flash] Error 1
> make[1]: Leaving directory '/home/jblair/git/USBaspLoader/firmware'
> make: *** [Makefile:14: flash] Error 2
> .
> .
> ======>BUILDING BOOTLOADER FIRMWARE
> .
> make[1]: Entering directory '/home/jblair/git/USBaspLoader/firmware'
> make[1]: Nothing to be done for 'all'.
> make[1]: Leaving directory '/home/jblair/git/USBaspLoader/firmware'
> make[1]: Entering directory '/home/jblair/git/USBaspLoader/firmware'
> .
> avrdude -c usbtiny -p atmega328p -U flash:w:main.hex:i
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.01s
> 
> avrdude: Device signature = 0x1e950f (probably m328p)
> avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
>          To disable this feature, specify the -D option.
> avrdude: erasing chip
> avrdude: reading input file "main.hex"
> avrdude: writing flash (30866 bytes):
> 
> Writing | ################################################## | 100% 0.00s
> 
> 
> avrdude: error: usbtiny_send: error sending control message: Connection timed out (expected 128, got -110)
> avrdude: 30866 bytes of flash written
> avrdude: verifying flash memory against main.hex:
> avrdude: load data flash data from input file main.hex:
> avrdude: input file main.hex contains 30866 bytes
> avrdude: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 0.00s
> 
> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x7000
>          0x69 != 0x6b
> avrdude: verification error; content mismatch
> 
> avrdude: safemode: Fuses OK (E:FC, H:C0, L:1F)
> 
> avrdude done.  Thank you.
> 
> make[1]: *** [Makefile:41: flash] Error 1
> make[1]: Leaving directory '/home/jblair/git/USBaspLoader/firmware'
> make: *** [Makefile:14: flash] Error 2
> ```

you can try usnoobie bootloader with your crystall: http://eleccelerator.com/usnoobie/downloads.php

Tenkir, 2020-12-02:
I was able to successfully flash the bootloader and the fuses. However, now when I plug the keyboard into my machine, it freezes all USB devices until unplugged.

Ardakilic, 2021-08-09:
I don't know if this would help but Googling brought me here several times when I have struggled.

I have bought ATMega328p from China, and wanted to flash the bootloader. Windows didn't help me at all with latest updates installed. macOS (and possibly all Linux variants would have) worked nicely instead.

I have installed the bootloader hex file successfully today with a Chinese generic USB ISP programmer, and within the help of avrdudess.

I wired like this:

[image]

I stole the commands from AVRDUDESS:

```bash
avrdude -c usbasp -p m328p -B 0.5 -V -U flash:w:"C:\full\path\main.hex":a -F
```
Then flashed like this in macOS:

[image]

I had no errors, and people say the warning is due to my ISP programmer being a Cheap Chinese one. (100nf capacitor on VCC line didn't help neither). I have not yet tried, but I believe this way it should work. It took a couple of seconds to upload.

I have downloaded the file like clicking to it, then clicking "view raw" , then saving as `file.hex`