hsgw/plaid #35: USB Device Unknown Descriptor after build

rwcts, 2021-07-26:
I have double and triple checked the soldering and looked at it under the scope but all looks good. After getting it all done I connected it via USB and I heard the Windows USB detection sound. That is when it showed that the device connected is having issues and gave an unknown descriptor message in device manager. Suggestions on what to do next?

jbeanland, 2021-07-28:
@rwcts any luck with this? I have the same thing but can only find old dead threads describing the problem, not any with a solution..

rwcts, 2021-07-28:
no luck as of yet. Since I have quadruple checked all solder joints I'm not sure. I'm thinking something may be soldered in wrong but all polarized components look to be correct according to the assembly instructions. There were some discrepancies in the assembly as the board I have is black and the picture is white. One of the assembly instructions i found show the chip to be installed one way but im following the board graphics to place the chip in correctly. I tried a different cable as that can definitely cause this type of issue too but I have used 4 cables with the same issue. Tried a different computer and port as that can cause it too. Otherwise no luck at this point.

hsgw, 2021-07-29:
Hi,
VUSB may have a compatibility problem with your computer's USB Host IC. 
Can you try it on another computer?

jbeanland, 2021-07-29:
no change for me :(

@hsgw what is a good series of steps for troubleshooting this?

I'm using windows, but I can also try with linux if needed.

The behaviour is: 
- If I plug it in (not pushing any buttons) then it is not recognised (USB device not recognised). Zadig shows it as unknown with no driver. It is also unknown with no driver in device manager.
- If I plug it in while holding boot then release, same behaviour. when I release the boot button, it comes up as not recognised.
- No change if I press any buttons after it is plugged in (like described here https://github.com/hsgw/plaid/blob/master/doc/en/bootloader.md#enter-bootloader-mode or any other combination)

Should I test the soldering with a multimeter? And what should I check?
I bought the kit from mechboards, so the chip should have the bootloader already flashed, but is there a way to test the chip and if the bootloader is working correctly?
What behaviour do I expect from the reset button and boot button?

Thanks!!

hsgw, 2021-07-29:
@jbeanland The following steps may be helpful.
- Verify that components with correct values are mounted in the correct locations 
- Use a multimeter to check if the soldering is done correctly
- If you have an Arduino or AVR writer, burn the bootloader again

If these steps completely confirm that it does not work, you know that it does not work due to component failure or error.

> but is there a way to test the chip and if the bootloader is working correctly?
There is no way to confirm this. The easiest way is simply to re-write bootloader. 

> What behavior do I expect from the reset button and boot button?
Reset is hardware reset, Boot is entering bootloader mode when the device starts.

rwcts, 2021-07-29:
I did attempt another computer with the same results. In the instructions on how to load a bootloader it should not detect a usb plugged in until clicking and holding down, in a various way the reset and power buttons. It detects the connection to the keyboard (unknown descriptor). Is there a way to start attempting to load the bootloader?

I did check continuity on all connections on the board. all are good too. I get voltage through parts of the board as well. 5v and around 3v (from memory)

hsgw, 2021-07-31:
> starting bootloader
There is no way to get into the bootloader other than by pressing the reset and boot buttons.
However, if there is no firmware written to it, it should automatically enter the bootloader.
This feature sometimes does not work well.

If you are sure that the hardware assembly is perfect and the bootloader has been written, then it might be a fault or error in the MCU, the crystal, or the capacitors around it.

VUSB requires a precise oscillator frequency, and some host computers may not work at all.
Therefore, it is very difficult for me to find out why your keyboard does not work.

jbeanland, 2021-07-31:
Thanks for the information @hsgw, this all helps a lot. I will try some stuff and report back if I find a solution.

benc3d, 2021-08-04:
Just wanted to say, I've also hit this issue.  Originally I sourced my own PCB's and components but after having this issue I then sourced a kit through mechboards and I'm still having the same issue.

A couple of things I did note, none of the LEDs light or blink at all, I'm not sure if they should? 

Also the BOOT / RESET buttons appear to be doing nothing.  If I hold both buttons down and plug in the USB, I still get the USB device not recognised prompt in windows and whilst plugged in, the reset button does nothing.  I assumed it should disconnect and reconnect if pressing the button with the USB connected? 

@rwcts "One of the assembly instructions I found show the chip to be installed one way but I'm following the board graphics to place the chip in correctly." which component was this?

rwcts, 2021-08-05:
> @rwcts "One of the assembly instructions I found show the chip to be installed one way but I'm following the board graphics to place the chip in correctly." which component was this?

It was the main chip on the board. It was showing the opposite direction in some instructions.

hsgw, 2021-08-06:
> @benc3d Also the BOOT / RESET buttons appear to be doing nothing.

The RESET button is a basic function of the MCU, so it is strange behavior that nothing happens.
If it is not properly recognized by the host computer, it may appear that nothing is happening on the host.

Just to clarify, after you press the RESET/BOOT button, do you release the RESET button first?

benc3d, 2021-08-06:
@hsgw correct, but I have tried all combinations of button presses :)  One thing I should have mentioned and I feel may be important is I have not soldered all my switches and diodes yet as I was wanting to test the MCU / main board before I soldered all my switches, but I've just checked the PCB and noticed the boot button is tied into the COL3 switch, so I will try that and hope I've not been wasting your time!

hsgw, 2021-08-07:
> @benc3d the boot button is tied into the COL3 switch

This is correct. BOOT and col3 share MCU pins.
And you can test without soldering diodes and switches.

benc3d, 2021-08-20:
@rwcts @jbeanland So after a bunch of debugging my issue turned out to be that my MCU that I purchased as part of the kit was not flashed with a bootloader (the description said it was).  I ended up following this guide to get the bootloader flashed https://www.algorist.co.uk/post/how-to-burn-the-plaid-keyboard-bootloader-using-arduino-nano/
(but I did it all on a bread board as I still wasn't sure if the issues was the MCU or a failed component) but after flashing the hex from that page, it all seems to be working!  I just need to finish soldering my switches now!