hsgw/plaid #36: Keyboard lag

richard-davey-arcadisgen, 2021-11-23:
I'm typing this up on a plaid and it's pretty hard going. There's lag in polling key presses, mostly after a period of not typing anything.

I've burned the bootloader myself using the standard fuse bit settings. It burns successfully and keymaps can be subsequently uploaded.

hsgw, 2021-11-30:
Hi, I have never had such a problem in my environment.
Since the USB device is emulated by software, it may be a limitation of the MCU's performance or something.
I'm sorry this isn't helping, just information.

richard-davey-arcadisgen, 2021-12-01:
My MCU is an ATMEL ATMEGA 328P U, which I'm assuming is what everyone else uses for this keyboard? The odd thing is that it is as if it takes a while to wake itself after a key press when left unattended for a while (around 30 seconds). This makes me think it is either a fuse bit setting or a firmware setting. I'm using a fairly up to date version of QMK (around June 2021).

richard-davey-arcadisgen, 2021-12-06:
Never mind. The keyboard is working now using the standard fuse bits and v-usb standard firmware and I'm pretty sure it must have been an external problem. Probably a loose cable, or extreme low temperatures or something.

closed 2021-12-06 by richard-davey-arcadisgen