hsgw/plaid #38: Switch plate compatibility (for hotswap)

YassineGherbi, 2022-03-03:
I know the Plaid Kit is shipped without a switch plate but I would like to know if there's any compatible plate (maybe with some modding). 

I've added Mill-Max sockets to made it hot-swappable but then discovered that this doesn't work all that well when you don't have a plate. I'm still pretty new to this hobby so If there's another way to make hot-swap work without a plate I would also appreciate it to be informed about those options. Thanks a lot!

hsgw, 2022-03-13:
I have seen someone making plate.
https://twitter.com/lekipon/status/1337675289464225794

YassineGherbi, 2022-03-20:
Any idea where it’s being sold? I’ve looked everywhere but can’t find anything.

thorstenweber83, 2022-04-20:
Looks like the USB-C fork of this repo also contains gerber files for a plate:
 https://github.com/piit79/plaid-c/blob/main/pcb/gerber/plaid-c-plate-1.2.42D.zip

You can use this file to order from jlcpcb.com or similar companies.