hsgw/plaid #39: Gerber files for main pcb, switch plate, bottom plate

syenasweta, 2023-05-29:
Where can I find it?

syenasweta, 2023-05-29:
Hi @hsgw and all developer of Plaid,

My capability is not in electronics. I just want to print and use your pcb for independent-use according to your original concept, namely through-hole with vusb.

So, would you mind making a gerber file (main pcb, switch plate pcb, and bottom pcb), so that ordinary people can use your ready-to-print master gerber file?

hsgw, 2023-06-04:
Different factories producing PCBs may require different Gerber data. I cannot guarantee that, so I do not distribute that directly.

And there are many helpful instruciton available, so download kicad and try it for yourself.

closed 2023-06-04 by hsgw